equinor/camille #107: Running tests with version of pytest >= 4.0.0 fails

lightenup, 2018-12-05:
The tests in Camille fails if they are run with pytest version 4.0.0 or above. This is most likely due to the 'repeat' library used in some of the tests.

Excrept from failure:
```E   _pytest.warning_types.RemovedInPytest4Warning: MarkInfo objects are deprecated as they contain merged marks which are hard to deal with correctly.
E   Please use node.get_closest_marker(name) or node.iter_markers(name).
E   Docs: https://docs.pytest.org/en/latest/mark.html#updating-code```

JensGM, 2018-12-08:
This is due to `pytest-repeat` not yet being compatible with `pytest 4.0.0`

JensGM, 2018-12-18:
This should be fixed later on

lightenup, 2018-12-18:
Good catch. I'm subscribing to new releases of pytest-repeat and can kickstart this issue when they have a new lease out.

PR #125 (2018-12-18): Locking pytest version until compatible pytest-repeat is available
Fixes #107
closed 2021-03-08 by JensGM